meltaxa/solariot #5: pv_power & total_pv_power dont seem mapped correctly

mvandersteen, 2019-01-19:
Hi

Just got your script working and was studying the data locally coming through and comparing with whats on screen. As a reference I have a battery attached to a Sungrow SH5K.

When looking at the data being pulled, after 9pm to battery draining, I can see the values for pv_power is the same as the battery power value, and the total_pv_power is 0.

I've changed the mapping for myself for now so that pv_power is called inverter_power and total_pv_power is changed to pv_power.

No big deal if you don't have a battery hooked up but just thought others might find this info handy.

BTW: Thanks for the script this is brilliant. I'm going to use this as the basis for a Home_assistant plugin if that's ok.

Cheers
Mark

mvandersteen, 2019-01-27:
No issue just personal preferences.

closed 2019-01-27 by mvandersteen

meltaxa, 2019-02-06:
Hi @mvandersteen. Feel free to use this in whatever form for HA. If you can wait a few weeks, I'll be releasing changes including your feedback to a develop branch. There will be a few fundamental changes to how Solariot is run and presents output which allow for better integration with HA.