meltaxa/solariot #9: Support for SB3.0-1AV-41 224

knights1, 2020-01-18:
Hi Meltaxa,

I have a SMA SB3.0-1AV-41 224 and am trying to get your solution to work.
Configuration towards influb works great but i need some help with the modbus config for my inverter.

do have any idea of the SB3.0-1AV-41 224 could work?

knights1, 2020-01-18:
config at the moment:

inverter_ip = "192.168.1.166"
inverter_port = 502
# Slave Defaults
# Sungrow: 0x01
# SMA: 3
model = "sma-SBn_n-1AV-40"
slave = 3
timeout = 3
scan_interval = 10
# Optional:
# dweepy_uuid = "random-uuid"
# Optional:
influxdb_ip = "192.168.1.91"
influxdb_port = 8086
influxdb_user = "xxxx"
influxdb_password = "xxxx"
influxdb_database = "inverter"
influxdb_ssl = False
influxdb_verify_ssl = False
# Optional
#mqtt_server = "192.168.1.91"
#mqtt_port = 1883
#mqtt_topic = "inverter/stats"
# See config-example.py for required and optional settings.

this gives me an error:

pi@raspberrypi:~/solariot/solariot-master $ ./solariot.py
Load config sma-SBn_n-1AV-40
Load ModbusTcpClient
Connect
[ERROR] fromRegisters() got an unexpected keyword argument 'endian'

meltaxa, 2020-02-03:
Your inverter model may not be supported by the contributed modbus map. However, try dropping the endian field from Line 137 to be:
`message = BinaryPayloadDecoder.fromRegisters(received.registers)`

crazycusti, 2020-02-04:
This doenst work, after this change i get wrong results.

try this:

message = BinaryPayloadDecoder.fromRegisters(received.registers, byteorder=Endian.Big, wordorder=Endian.Big)

in pymodbus 1.4.0 they changed the endian-stuff a little bit.

looks good for me

root@rietjenserver:/home/rietjen/solariot-master# python3 solariot.py 
Load config sma-SBn_n-1AV-40
Load ModbusTcpClient
Connect
{'30531 - Total yield (kWh)': 227, '30535 - Daily yield (Wh)': None, '30537 - Daily yield (kWh)': None, '30769 - DC current input [1] (A)': None, '30771 - DC voltage input [1] (V)': None, '30775 - Power (W)': None, '30783 - Grid voltage phase L1 (V)': None, '00000 - Timestamp': '2020-02-04 22:01:04'}
[INFO] Sent to InfluxDB

before, i got extreme high results (68000 watts in the night, yay!)

knights1, 2020-02-04:
I cannot start with python3?

pi@raspberrypi:~/solariot/solariot-master $ python3 solariot.py

Traceback (most recent call last):

  File "solariot.py", line 27, in <module>

    import paho.mqtt.client as mqtt

ModuleNotFoundError: No module named 'paho'

knights1, 2020-02-04:
By the way: i am looking forma way to get the daily yield. In your
information there in no daily yield. Did you you yield today?

crazycusti, 2020-02-05:
the adress is 30537

i started the logger again with a clean database in the night, so the sma modbus do not transmit DC related things. When the sun go up/rise/myenglishistoobad the inverter started and sent these datas.

when you use python3, you must install everything with pip3, not pip. this is a littlebit confusing, python is a version-chaos. python2 and python3 are complete different spaces. afaik you can use python2, there is no must-to-have python3 things in this code. 

also use python instead of python3.

crazycusti, 2020-02-05:
oh nvm, 30357 is without data.. sorry.

i look for a fix, this evening.

closed 2020-09-14 by meltaxa