meltaxa/solariot #10: Compatibility with SG6K-D inverter

brunolinsalves, 2020-01-31:
Hi, 

I have a sungrow inverter (model SG6K-D) and I want to know if I can use this script (with modbus-sungrow-sh5k.py values). 
I don't know if I could explain, but how do I use this script with SG6K-D inverter ? Can I use the config.py with model = "sungrow-sh5k" ?

Thanks

brunolinsalves, 2020-02-02:
I tried to execute with this config.py and got the [ERROR] 'ModbusIOException' object has no attribute 'registers'.  The IP of the inverter is correct. 

```
inverter_ip = "192.168.129.173"
inverter_port = 502
# Slave Defaults
# Sungrow: 0x01
# SMA: 3
slave = 0x01
model = "sungrow-sh5k"
timeout = 3
scan_interval = 10
```

meltaxa, 2020-02-03:
Check your slave address is correct too. If you do not know it, try cycling through different addresses from 0x01 to 0x04.

mpp-piotrp, 2020-03-28:
I have the same problem with sungrow-sh5k after a remote firmware upgrade. Have they changed the Modbus map? Slave address?

chrisvivo, 2020-04-07:
I have the same issue for a month or so after firmware.. have tried different slave address etc, no luck.

Otherwise, it's been running great and thank you!

meltaxa, 2020-04-10:
Unfortunately (or perhaps fortunately), I have an older wifi module that wasn't in the firmware update scope provided by Sungrow. 

Can you confirm the default TCP port (502) is still open?

chrisvivo, 2020-04-11:
Thanks for looking, yep, 502 is open:

Open TCP Port: 	80     		http 
Open TCP Port: 	502    		asa-appl-proto

mopedy, 2020-07-14:
Hi meltaxa,

thanks for your great work, was working perfectly until recently. I guess i have same problem as chrisvivo.
Did you get it to work again?

chrisvivo, 2020-07-14:
Unfortunately not, I tried a multitude of things within my limited ability.

mopedy, 2020-07-18:
I managed to get it working again, by using a old wlan module like you described above. Interestingly i was using it over Ethernet and not Wifi before, by setting up the wifi Module on the inverter again, it somehow opened the ethernet cable port as well.  Which allows me to use my old setup without the wifi module again.

rpvelloso, 2020-09-09:
I have a SG4K with an wi-fi interface. Tried using it with your SH5K config, got this:
Load config sungrow-sg4k
Load ModbusTcpClient
Connect
[ERROR] 'ModbusIOException' object has no attribute 'registers'
[ERROR] 'ModbusIOException' object has no attribute 'registers'
[ERROR] 'ModbusIOException' object has no attribute 'registers'
{}

Any idea? Where can I find register mapping for my inverter? How did you find yours? 

Thanks.

meltaxa, 2020-09-14:
Refer to #12 which will track any updates on using the SungrowModbusTcpClient library which addresses the encryption issue on later firmware versions.

closed 2020-09-14 by meltaxa