meltaxa/solariot #11: UTF-8 error

crazycusti, 2020-02-03:
SMA SB3.6-1AV-41 Firmware 3.0.7R + Python 2.7/3.7 (Debian 10 Buster)

`Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "solariot.py", line 177, in publish_influx
    target=flux_client.write_points([metrics])
  File "/usr/local/lib/python2.7/dist-packages/influxdb/client.py", line 530, in write_points
    consistency=consistency)
  File "/usr/local/lib/python2.7/dist-packages/influxdb/client.py", line 598, in _write_points
    protocol=protocol
  File "/usr/local/lib/python2.7/dist-packages/influxdb/client.py", line 331, in write
    data = make_lines(data, precision).encode('utf-8')
  File "/usr/local/lib/python2.7/dist-packages/influxdb/line_protocol.py", line 158, in make_lines
    value = _escape_value(field_value)
  File "/usr/local/lib/python2.7/dist-packages/influxdb/line_protocol.py", line 94, in _escape_value
    value = _get_unicode(value)
  File "/usr/local/lib/python2.7/dist-packages/influxdb/line_protocol.py", line 109, in _get_unicode
    return data.decode('utf-8')
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xff in position 24: invalid start byte
`

Modbus Info:
`{'30205 - Nominal power in Warning Mode (W)': 0, '40210 - Operating mode of feed-in management': 14084, '40464 - Voltage monitoring of median minimum threshold (V)': 379.05, '30833 - cosPhi excit.type, cosPhi config., direct spec.': 4612, '41023 - Number of points to be used, conf. of grid integr. char. 1': None, '41223 - Fallback react power Q in % of WMax for absent react power ctr (%)': 0.0, '40230 - Act. power at start point, cosPhi(P) char. config. (%)': 12800, '30769 - DC current input [1] (A)': 8388.608, '41063 - 2nd activation of the characteristic curve, configuration of characteristic curve mode': 4278255103, '40434 - Frq. monitoring lower max. threshold trip. time (ms)': 51200, '30517 - Daily yield (Wh)': 18446744073709551615L, '30541 - Operating time (s)': 704699974, '40200 - Operating mode of stat.V stab., stat.V stab. config.': 12804, '43090 - Login with Grid Guard-Code': 0, '30837 - Active power limitation P, active power configuration (W)': 24590, '40185 - Maximum apparent power device (VA)': 24590, '40183 - Phase assignment': 17665, '30561 - Number of events for installer': 1536, '30535 - Daily yield (Wh)': None, '41017 - Adjustment time of characteristic operating point, conf. of grid integr. char. 1 (s)': None, '41075 - Output frequency, conf. of grid integration char. 2': 4278255103, '30783 - Grid voltage phase L1 (V)': None, '30953 - Internal temperature (C)': 8388608, '30211 - Recommended action': 30467, '30865 - Power grid reference (W)': 43269, '40204 - Reactive power set value as a % (%)': 0.0, '30537 - Daily yield (kWh)': None, '30203 - Nominal power in Ok Mode (W)': 24590, '41035 - Y value 2, conf. of grid integr. char. 1': 8388.608, '35385 - Number of events for service': 14848, '41255 - Normalized active power limitation by PV system ctrl (%)': 9.75, '41085 - X value 3, conf. of grid integr. char. 2': 8388.608, '30927 - Duplex mode of SMACOM A': 48902, '40442 - Frq. monitoring median min. threshold trip. time (ms)': None, '30599 - Number of grid connections': 44288, '40009 - Operating condition': 9985, '30867 - Power grid feed-in (W)': 0, '30231 - Maximum active power device (W)': 24590, '30829 - Reactive power set value as a % (%)': 0.0, '41227 - Timeout for absent cos Phi spec (s)': 22530, '41219 - Operating mode for absent reactive power control': 51721, '30053 - Device type': 47652, '30543 - Feed-in time (s)': 620772904, '41195 - Timeout for absent active power limitation (s)': 22530, '31033 - Current speedwire gateway address': '192.168.178.1\x00\x00\x00', '40513 - Speedwire DNS server address': '192.168.1.1\x00\x00\x00\x00\x00', '40452 - Voltage monitoring lower maximum threshold (V)': 210.95, '40466 - Voltage monitoring median min. threshold trip.time (ms)': 4135, '41171 - Set total yield (kWh)': 0, '41123 - Min. voltage for reconnection (V)': 241.4, '41077 - X value 1, conf. of grid integr. char. 2': 8388.608, '30929 - Speedwire connection status of SMACOM A': 13057, '40484 - Activation of active power gradient': 13313, '30199 - Waiting time until feed-in (s)': None, '40240 - Activation of stay-set indicator function, linear instantaneous power gradient configuration': 12033, '40224 - Excit. type at start point, cosPhi(P) char. conf.': 4356, '30975 - Intermediate circuit voltage (V)': 83886.08, '41197 - Fallback act power lmt P in % of WMax for absent act power lmt (%)': 41.35, '30835 - Operating mode of feed-in management': 14084, '41027 - Y-axes reference, conf. of grid integration char. 1': 4278255103, '41025 - X-axes reference, conf. of grid integration char. 1': 4278255103, '30813 - Apparent power (VA)': 8388608, '30525 - Feed-in time (s)': 620772904, '41087 - Y value 3, conf. of grid integr. char. 2': 8388.608, '40159 - Speedwire IP address': '192.168.1.170\x00\x00\x00', '31017 - Current speedwire IP address': '192.168.178.21\x00\x00', '40226 - cosPhi at end point, cosPhi(P) char. config.': 256.0, '41229 - Fallback cos Phi for absent cos Phi spec': 4135, '31085 - Nominal power in Ok Mode (W)': 24590, '41033 - X value 2, conf. of grid integr. char. 1': 8388.608, '30831 - cosPhi setpoint, cosPhi config., direct specif.': 230.4, '40206 - cosPhi setpoint, cosPhi config., direct specif.': 230.4, '34113 - Internal temperature (C)': 8388608, '40631 - Device name': 'SB3.6-1AV-41 573\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff', '30207 - Nominal power in Fault Mode (W)': 0, '30051 - Device class': 16671, '30775 - Power (W)': 8388608, '41061 - 2nd characteristic curve number, configuration of characteristic curve mode': None, '30213 - Message': 30211, '41081 - X value 2, conf. of grid integr. char. 2': 8388.608, '31041 - Current speedwire DNS server address': '192.168.178.1\x00\x00\x00', '30059 - Software package': 197639, '30977 - Grid current phase L1 (A)': 8388.608, '41225 - Operating mode for absent cos Phi spec': 51721, '40497 - MAC address': '00:40:AD:AA:DA:E7\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', '40448 - Voltage monitoring median maximum threshold (V)': 246.85, '30787 - Grid voltage phase L3 (V)': None, '40432 - Frequency monitoring lower maximum threshold (Hz)': 77.0, '41037 - X value 3, conf. of grid integr. char. 1': 8388.608, '41253 - Fast shut-down': 47877, '41111 - Voltage monitoring of lower minimum threshold as RMS value (V)': 379.05, '40157 - Automatic speedwire configureation switched on': 26884, '41129 - Upper frequency for reconnection (Hz)': 361.15, '40440 - Frequency monitoring median minimum threshold (Hz)': None, '35377 - Number of events for user': 21249, '40238 - Active power gradient, linear instantaneous power gradient configuration (%)': 10240, '40428 - Frequency monitoring median maximum threshold (Hz)': None, '31247 - Residual current (A)': 8388.608, '41041 - X value 4, conf. of grid integr. char. 1': 8388.608, '40208 - cosPhi excit.type, cosPhi config., direct spec.': 4612, '41221 - Timeout for absent reactive power control (s)': 22530, '30925 - Connection speed of SMACOM A': 47366, '41117 - Voltage monitoring of upper max. thresh. as RMS value for tripping time (ms)': 24810, '41203 - Nominal PV system power (W)': 24590, '41073 - Input unit, conf. of grid integration char. 2': 4278255103, '41029 - X value 1, conf. of grid integr. char. 1': 8388.608, '30773 - DC power input [1] (W)': 8388608, '40167 - Speedwire subnet mask': '255.255.255.0\x00\x00\x00', '40458 - Voltage monitoring lower minimum threshold (V)': 574.15, '40436 - Frequency monitoring upper minimum threshold (Hz)': 363.7, '41083 - Y value 2, conf. of grid integr. char. 2': 8388.608, '40005 - Standard/Daylight saving time conversion on': 26884, '40430 - Frq. monitoring median max. threshold trip. time (ms)': None, '40456 - Voltage monitoring lower max. threshold trip. time (ms)': 51200, '40175 - Speedwire gateway address': '192.168.1.1\x00\x00\x00\x00\x00', '30215 - Fault correction measure': 29955, '40470 - Island network detect. status': 4278255103, '40003 - Time zone': 27173, '40220 - Difference between reset frequency and grid frequency, linear instanta [... cut ...]  phase L2 (V)': None, '40067 - Serial number': 498290093, '40214 - Active power limitation P, active power configuration (%)': 25600}
`

my config:

`inverter_ip = "192.168.178.21"
inverter_port = 502
# Slave Defaults
# Sungrow: 0x01
# SMA: 3
slave = 3
model = "sma-SBn_n-1AV-40"
timeout = 3
scan_interval = 10
# Optional:
influxdb_ip = "******"
influxdb_port = 8086
influxdb_user = "*****"
influxdb_password = "****"
influxdb_database = "inverter3"
influxdb_ssl = False
influxdb_verify_ssl = False
`

crazycusti, 2020-02-04:
i commented out the useless data from modbus-sma-SBn_n-1AV-40.py

  ['30531 - Total yield (kWh)',30531,'U32','FIX0'],
  ['30533 - Total yield (MWh)',30533,'U32','FIX0'],
  ['30535 - Daily yield (Wh)',30535,'U32','FIX0'],
  ['30537 - Daily yield (kWh)',30537,'U32','FIX0'],
  ['30769 - DC current input [1] (A)',30769,'S32','FIX3'],
  ['30771 - DC voltage input [1] (V)',30771,'S32','FIX2'],
  ['30773 - DC power input [1] (W)',30773,'S32','FIX0'],
  ['30775 - Power (W)',30775,'S32','FIX0'],
  ['30783 - Grid voltage phase L1 (V)',30783,'U32','FIX2'],
  ['30953 - Internal temperature (C)',30953,'S32','TEMP'],

is enough for me.

work perfect with influxdb 1.x, python 3.7.x and a SMA SB3.6-1AV-41 Firmware 3.0.7R

:)

meltaxa, 2020-02-04:
Looks like there are slight differences between models / versions and modbus maps. Good work on getting the data you need.

closed 2020-02-04 by meltaxa