meltaxa/solariot #15: ValueError: AES key must be either, 24, or 32 bytes lon 16g

wombatvic, 2020-10-17:
Environment: Sungrow Inverter SG8K-D with Wifi Adaptor. Raspi8 with Python 3.8. 
Port 502 accessible by scanning. 
Error: ValueError: AES key must be either, 24, or 32 bytes lon 16g

I have attempted to re-install and update all dependencies with no errors. 

./solariot.py
Load config sungrow-sh5k
Load SungrowModbusTcpClient
Connect
Traceback (most recent call last):
  File "./solariot.py", line 77, in <module>
    client.connect()
  File "/solariot/SungrowModbusTcpClient.py", line 32, in connect
    self._getkey()
  File "/solariot/SungrowModbusTcpClient.py", line 22, in _getkey
    self._aes_ecb = AES.new(self._key, AES.MODE_ECB)
  File "/usr/lib/python2.7/dist-packages/Crypto/Cipher/AES.py", line 94, in new
    return AESCipher(key, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/Crypto/Cipher/AES.py", line 59, in __init__
    blockalgo.BlockAlgo.__init__(self, _AES, key, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/Crypto/Cipher/blockalgo.py", line 141, in __init__
    self._cipher = factory.new(key, *args, **kwargs)
**ValueError: AES key must be either, 24, or 32 bytes lon 16g**

inverter_ip = "10.1.1.7"
inverter_port = 502
# Slave Defaults
# Sungrow: 0x01
# SMA: 3
slave = 0x01
model = "sungrow-sh5k"
timeout = 3
scan_interval = 10
# Optional:
dweepy_uuid = "random_solar_thing"
# Optional:
influxdb_ip = "192.168.1.128"
influxdb_port = 8086
influxdb_user = "user"
influxdb_password = "password"
influxdb_database = "inverter"
influxdb_ssl = True
influxdb_verify_ssl = False
# Optional
mqtt_server = "192.168.1.128"
mqtt_port = 1883
mqtt_topic = "inverter/stats"
mqtt_username = "user"
mqtt_password = "password"

meltaxa, 2020-10-17:
Hi @wombatvic. According to output, it seems the Python runtime is 2.7. Try running in version 3. Using a Python3 virtual environment (venv) will ensure you use the required version libraries.

wombatvic, 2020-10-18:
Thanks for that - using the venv worked and got me past the encryption error. 

Have I missed a step with the below error querying the registers? I'm using the SG8K-D with Wifi Adaptor.

Load config sungrow-sh5k
Load SungrowModbusTcpClient
Connect
[ERROR] 'ExceptionResponse' object has no attribute 'registers'
[ERROR] 'grid_import_or_export'

meltaxa, 2020-10-22:
Hey @wombatvic, glad you got past the encryption issue. I was hoping someone in the community might be able to assist with your particular inverter model. I am unable to verify if the SH5K modbus map is compatible with the SG8K-D. I'll leave this ticket open in the hope someone can contribute their findings.

michael-robbins, 2021-02-23:
Hey @wombatvic if you're still having this problem try out the sungrow-sg5kd modbus map that's in there now, might be a more compatible fit?

meltaxa, 2021-03-25:
Bump @wombatvic. How'd you go with your SG8KD using a SG5KD map? There are others interested in the compatibility.

shannonpasto, 2021-05-01:
I came across http://www.photovoltaicsolar.in/Sungrow_Manual/Modbus%20RS485%20RTU%20Protocol.pdf in my travels. It doesn't explicitly mention the SG8KD but you might be able to use trial and error to get it working. I have an SG3KD and while the SG5KD map worked I needed to modify it a bit to remove unused/incorrect registers. Using the link helped me understand what the register ranges are

meltaxa, 2021-05-02:
Thanks @shannonpasto, that's a great find. Would you mind sharing your sg3kd map? I'll add it to repo.

michael-robbins, 2021-05-02:
I just used https://solarclarity.co.uk/wp-content/uploads//2018/12/TI_20180301_String-Inverters_Communication-Protocol_V10_EN.pdf to validate a few things on my SG5K-D and they seemed to line up quite nicely as well!

shannonpasto, 2021-05-03:
> Thanks @shannonpasto, that's a great find. Would you mind sharing your sg3kd map? I'll add it to repo.

Here it is. Sorry if the formatting didn't work out properly

```
read_register = {
  "5003": "daily_power_yield_0.01", # Wh
  "5004": "total_power_yield_100",  # MWh
  "5008": "internal_temp_10",       # C
  "5011": "pv1_voltage_10",         # V
  "5012": "pv1_current_10",         # A
  "5013": "pv2_voltage_10",         # V
  "5014": "pv2_current_10",         # A
  "5017": "total_pv_power",         # W
  "5019": "grid_voltage_10",        # V
  "5022": "inverter_current_10",    # A
  "5031": "total_active_power",     # W
  "5036": "grid_frequency_10"      # Hz
}

holding_register = {
  "5000": "year",
  "5001": "month",
  "5002": "day",
  "5003": "hour",
  "5004": "minute",
  "5005": "second"
}

scan = """{
    "read": [
        {
            "start": "5000",
            "range": "100"
        },
        {
            "start": "5100",
            "range": "50"
        }
  ],
  "holding": [
    {
      "start": "4999",
      "range": "10"
    }
  ]
}"""

# Match Modbus registers to pvoutput api fields
# Reference: https://pvoutput.org/help/api_specification.html#add-status-service
pvoutput = {
  "Energy Generation": "daily_power_yield",
  "Power Generation": "total_active_power",
  "Temperature": "internal_temp",
  "Voltage": "grid_voltage"
}```

michael-robbins, 2021-05-10:
@meltaxa maybe we just close this off if @wombatvic isn't responding?

closed 2021-05-10 by meltaxa