meltaxa/solariot #16: SG5K-D  No data extracted

plague-doctor, 2020-10-27:
```
nmap -p- 172.16.50.116

PORT    STATE SERVICE
80/tcp  open  http
502/tcp open  mbap
```
cat config.py
```
inverter_ip = "172.16.50.116"
inverter_port = 502
# Slave Defaults
# Sungrow: 0x01
# SMA: 3
slave = 0x01
model = "sungrow-sh5k"
timeout = 5
scan_interval = 30
# Optional:
dweepy_uuid = "random-uuid"
# Optional:
influxdb_ip = "192.168.1.128"
influxdb_port = 8086
influxdb_user = "user"
influxdb_password = "password"
influxdb_database = "inverter"
influxdb_ssl = True
influxdb_verify_ssl = False
# Optional
mqtt_server = "192.168.1.128"
mqtt_port = 1883
mqtt_topic = "inverter/stats"
mqtt_username = "user"
mqtt_password = "password"
```
Returns:

```
./solariot.py                 
Load config sungrow-sh5k
Load SungrowModbusTcpClient
Connect
[WARN] No data. Try increasing the timeout or scan interval.
[WARN] No data. Try increasing the timeout or scan interval.
[WARN] No data. Try increasing the timeout or scan interval.
```

I have tried to increase timeout and scan_interval but is gives same output...

I am happy to help investigate but not sure where to start...

scottah, 2020-11-15:
get anywhere? I've got a SG8K-D and have the same output as yourself.

scottah, 2020-11-15:
Might be worth noting that I couldn't get pycrypto installed properly, so have resorted to pycryptodome - which is a drop-in replacement.

michael-robbins, 2021-02-07:
This is what I'm doing atm to get my `SG5K-D` running, I'll be PR'ing it into this repo once my other PRs land!

```
root@b3b2a4bfddac:/solariot# cat modbus-sungrow-sg5kd.py
read_register = {
  "5003":  "daily_power_yield_10",
  "5004":  "total_power_yield",
  "5008":  "internal_temp_10",
  "5011":  "pv1_voltage_10",
  "5012":  "pv1_current_10",
  "5013":  "pv2_voltage_10",
  "5014":  "pv2_current_10",
  "5017":  "total_pv_power",
  "5019":  "grid_voltage_10",
  "5022":  "inverter_current_10",
  "5031":  "total_active_power",
  "5036":  "grid_frequency_10",
}
holding_register = {
  "5000": "year",
  "5001": "month",
  "5002": "day",
  "5003": "hour",
  "5004": "minute",
  "5005": "second"
}
scan = """{
  "read": [
     {
       "start": "5000",
       "range": "40"
     }
  ],
  "holding": [
    {
       "start": "4999",
       "range": "100"
    }
  ]
}"""
```

michael-robbins, 2021-02-07:
Currently going through the registers and matching it all up against my isolarcloud app thingy!

michael-robbins, 2021-02-21:
I've created PR #20 which at least for me is working!

meltaxa, 2021-02-22:
PR#20 has been merged. Thanks for the contribution.

closed 2021-02-22 by meltaxa