meltaxa/solariot #22: PVOutput support

meltaxa, 2021-02-24:
Enhancement to upload Solariot metrics to pvoutput.org.

API specs: https://pvoutput.org/help.html#api-spec

michael-robbins, 2021-02-24:
PR's up, just needs testing once my solar inverter is online again... why can't the sun just stay up all the time...

michael-robbins, 2021-02-27:
Tested it out in my PR and it all seems to be working and publishing nicely!

Looking at `modbus-sungrow-sh5k.py` I can't see any of the `v1` through `v4` attributes though, are they different names?

Currently I've got:
```
            if "daily_power_yield" in metrics:
                parameters["v1"] = metrics["daily_power_yield"]

            if "total_active_power" in metrics:
                parameters["v2"] = metrics["total_active_power"]

            if "daily_energy_consumption" in metrics:
                parameters["v3"] = metrics["daily_energy_consumption"]

            if "power_meter" in metrics:
                parameters["v4"] = metrics["power_meter"]
```

meltaxa, 2021-03-06:
As per comments in your PR, I've made some updates to handle different modbus maps and also stay within the API rate limits.

PR #26 (2021-03-06): Bringing in pvoutput publishing and minor improvements
This was pulled from https://github.com/corvax/solariot/commit/d4146205acde600233eca1d967c86178337bda18
So credits to @corvax for the hard work!
I've just tweaked it by wrapping it in a class to make it a bit more modular/portable as well as expanding on the other `vX` parameters you can provide to pvoutput!

Additional features as I was doing this all at once:
* Modbus maps can now have `metric_100` or `metric_10000` to have the script divide by that figure to get you to the target decimal (this can even be say... `metric_0.01` to move th [... cut]
closed 2021-03-06 by meltaxa