meltaxa/solariot #27: Modbus mappings for S100 power meter 'history'

michael-robbins, 2021-03-01:
So for my Sungrow SG5K-D + S100 energy meter when the solar is low enough, the inverter shuts off and the modbus cannot be scraped anymore (Wifi dongle shuts off with it).

But when the inverter turns back on every morning when the sun rises the iSolarcloud app magically has all the history of the overnight power usage of the house.

So that means (most likely?) that there's possibly more modbus read registers that contain this? Or would this be some other sort of proprietary storage/replay of the overnight data?

Wondering if there's some sort of way to read this data out and 'replay' it to any of our other external integrations (mqtt/pvoutput/etc)

michael-robbins, 2021-03-01:
Worth mentioning is that the SG5K-D is a non-hybrid inverter so there's no battery, as I'd assume inverters integrated with a battery won't ever shut off? And be online 24/7 for querying?

meltaxa, 2021-03-01:
Sounds like the inverters software is doing the magic of caching and releasing the telemetry to iSolarcloud. I suspect this would be outside of the modbus protocols.  I have used Telegraf to collect (it can do modbus inpirts) and send my telemetry (to InfluxDB) which has the ability to cache when the outgoing connection is disrupted. Perhaps the inverter has a similar service or process running on it.

How is the inverter connecting to iSolarcloud? Probably https. If it's http/ftp, wireshark the payloads.

The SH5K is a hybrid (attached to a battery) and indeed always on.

michael-robbins, 2021-03-01:
Yeah it's https so no dice there.

[image]

Last point here says the power meter itself keeps the history, so my guess is the inverter at startup 'downloads' the overnight history and submits it separately to isolarcloud, probably skipping making it available to the modbus entirely.

I guess if this is the case then we won't be able to extract this info from the inverter/modbus registers... Maybe will look into isolarcloud website scraping to extract this... 

Anyway, will close this off, don't think there's anything to do here!

closed 2021-03-01 by michael-robbins

meltaxa, 2021-03-01:
See my (now defunct) [Herculis](https://github.com/meltaxa/herculis) project which scraped iSolarcloud for a downloadable csv file, before they redesigned it. Although it doesn't work anymore, it might give you some useful pointers.

joemoco7, 2022-01-07:
Sorry for the off-topic question, but you guys seem to know what you're talking about. 

I have a Sungrow SH5K-30 that I'm trying my darndest to get talking with a DTSU666 meter. My current guess is that the DTSU666's communication settings aren't configured properly - you can change parameters for the baud rate (4800 or 9600), communication stop bit and parity bits (none parity, 2 stop bits; none parity, 1 stop bit; even parity, 1 stop bit; odd parity, 1 stop bit), and communication address. 

Knowing that the inverter was happily talking with a S100 just before, I wondered if I can just set the DTSU666 to be the same. Would you happen to know what that might be? 😅 Thanks a ton for your help!

michael-robbins, 2022-01-07:
I did find this https://s3-ap-southeast-2.amazonaws.com/solarbrain/uploads/attachment/file/410/0dcc24414d19835010a890679d50c468.pdf that seems to mention a few things like baud rate/etc, not sure if that helps, but good luck!

joemoco7, 2022-01-07:
Thanks a ton for replying! I appreciate you looking into it enough to even find that manual 😃 Yeah, I had come across that one before - One of the non-Sungrow DTSU666 manuals out there kept mentioning "ModBus RTU," so earching Google for "S100 modbus" brought me to your forum 😅.

I know fairly confidently that the baud rate should be 9600, and if I knew one of the other two parameters, I might be able to just trail-and-error cycle through settings for the last parameter.

I did come across someone who took readings off a S100, and they got this:
Modbus Response (COM3)
Address: 32
Function: 3 (0x03) – Read Holding Registers
Byte Count: 4
Values: 00 00 1f b4
Register0: 0
Register1: 8116
Checksum: 50038(OK)

That "Address: 32" looked hopeful, but I'm not all that sure if that's what I'm looking for 😅.