meltaxa/solariot #28: Github Actions integration

michael-robbins, 2021-03-01:
If I was to PR a Github Actions file that built and published a docker image... how would that sound?

We can publish either to githubs docker image registry or to docker hub as well (or both!)

michael-robbins, 2021-03-01:
@meltaxa I'm not across publishing docker images to githubs registry, can suss out what you need to do/turn on/etc...

For docker hub if you want to be the owner, just go in and register and create a solariot and we'll throw the API KEY in as a secret variable into the repo

meltaxa, 2021-03-01:
Sounds good. Let's do both. I'll ping you the API keys when they are setup.

michael-robbins, 2021-03-01:
Ahh mad, yeah I was using that integration before I moved it all to Github Actions for a side project of mine.

We can just stick with this until there's a burning need to use something else I guess, less fiddly bits!

michael-robbins, 2021-03-01:
I think just master/latest is fine, as I build images separately for my branches when testing out stuff

closed 2021-03-01 by michael-robbins

meltaxa, 2021-03-06:
Ended up using Github Actions so we could have ARM platform images too. The Docker images works fine on my Raspberry Pi.