meltaxa/solariot #37: ERROR:pymodbus.factory:unpack requires a buffer of 4 bytes

erikderzweite, 2021-04-25:
Hello! I use the solariot with Sungrow SH6-0RT inverter and mqtt. Inverter and raspberry are in the same LAN, mqtt server ist remote.

Works fine for some time, I can get messages from mqtt. Then the script is stuck in the following error loop:

```
ERROR:pymodbus.factory:unpack requires a buffer of 4 bytes
ERROR:pymodbus.transaction:Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/ubuntu/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 207, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/ubuntu/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
WARNING:root:Modbus connection failed
WARNING:root:Failed to scrape inverter, sleeping until next scan
```

My config.py:
```
inverter_ip = "192.168.178.125"
inverter_port = 502
# Slave Defaults
# Sungrow: 0x01
# SMA: 3
slave = 0x01
model = "sungrow-sh5k"
timeout = 3
scan_interval = 10

# Optional:
dweepy_uuid = "random-uuid"

# Optional
mqtt_server = "mqtt.domain.tld"
mqtt_port = 1883
mqtt_topic = "inverter/stats"
mqtt_username = "xxx"
mqtt_password = "xxx"
```

shannonpasto, 2021-04-28:
I had a similar issue. Uninstalling latest SungrowModbusTcpClient and pymodbus modules and installing the versions needed in the requirements file seems to have removed the errors. Now I'm getting...

`WARNING:root:Modbus connection failed`
`WARNING:root:Failed to scrape inverter, sleeping until next scan`

meltaxa, 2021-05-01:
To update libraries, you can also run this command:

`pip install --upgrade -r requirements.txt`

Try the dev branch which has the dependency updates. Let me know how you go and I'll merge.

meltaxa, 2021-05-02:
Also run with verbose logging enabled. You can increase verbose logging by adding more v's, for example -vv or -vvv.

`solariot.py -v`

bdog720, 2021-05-08:
Similar issue (running dev branch)

ERROR:pymodbus.factory:index out of range
ERROR:pymodbus.transaction:Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\site-packages\pymodbus\transaction.py", line 207, in execute
    self.client.framer.processIncomingPacket(response,
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\site-packages\pymodbus\framer\socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\site-packages\pymodbus\framer\socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
WARNING:root:Modbus connection failed
WARNING:root:Failed to scrape inverter, sleeping until next scan
Traceback (most recent call last):
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\Users\Administrator\.vscode\extensions\ms-python.python-2021.4.765268190\pythonFiles\lib\python\debugpy\__main__.py", line 45, in <module>
    cli.main()
  File "c:\Users\Administrator\.vscode\extensions\ms-python.python-2021.4.765268190\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 444, in main
    run()
  File "c:\Users\Administrator\.vscode\extensions\ms-python.python-2021.4.765268190\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 285, in run_file
    runpy.run_path(target_as_str, run_name=compat.force_str("__main__"))
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 268, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\solariot\solariot.py", line 490, in <module>
    time.sleep(config.scan_interval)
    
    
I will try increase verbose logging to get more detail

bdog720, 2021-05-08:
This was with -vvv

Not sure if no response from inverter is expected or not? However once this happens it never fixes itself. Need to restart the script and it works straight away

DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.249', 54331)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.249', 54332)
DEBUG:pymodbus.transaction:SEND: 0x7 0x1c 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x88 0x0 0x64
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received))
DEBUG:pymodbus.framer.socket_framer:Processing:
DEBUG:pymodbus.transaction:Getting transaction 1820
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
WARNING:root:Modbus connection failed
WARNING:root:Failed to scrape inverter, sleeping until next scan

michael-robbins, 2021-05-10:
Only thing that comes to mind is my inverter didn't like too many connections at once, and occasionally it'd just disconnect/ignore new solariot scripts polling it.

Killing the script and running it again a few times would eventually get me a correct response from the inverter.

Are you always seeing these?
```
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received))
```

Or just occasionally? but still getting the occasional valid response/payload of data

bdog720, 2021-05-11:
I'll play with timeout and scan_interval to see if a longer wait fixes the issue. Will create a new issue if it persists

meltaxa, 2021-09-03:
Also, try the latest code which addresses a pymodbus version issue.

closed 2022-01-26 by meltaxa