meltaxa/solariot #38: Error during required libraries install

bdog720, 2021-04-29:
Sorry I'm not experienced with Python so not sure what else to add but this is the full output

C:\Users\Administrator\Documents\Solariot>pip install -r requirements.txt
WARNING: Value for scheme.headers does not match. Please report this to <https://github.com/pypa/pip/issues/9617>
distutils: c:\users\administrator\appdata\local\programs\python\python39\Include\UNKNOWN
sysconfig: c:\users\administrator\appdata\local\programs\python\python39\Include
WARNING: Additional context:
user = False
home = None
root = None
prefix = None
Collecting paho-mqtt>=1.4.0
  Using cached paho-mqtt-1.5.1.tar.gz (101 kB)
Requirement already satisfied: pymodbus>=1.3.2 in c:\users\administrator\appdata\local\programs\python\python39\lib\site-packages (from -r requirements.txt (line 2)) (2.5.1)
Collecting dweepy>=0.3.0
  Using cached dweepy-0.3.0.tar.gz (8.1 kB)
Collecting influxdb>=2.17.0
  Using cached influxdb-5.3.1-py2.py3-none-any.whl (77 kB)
Requirement already satisfied: python_dateutil>=2.6.1 in c:\users\administrator\appdata\local\programs\python\python39\lib\site-packages (from -r requirements.txt (line 5)) (2.8.1)
Requirement already satisfied: pytz>=2017.2 in c:\users\administrator\appdata\local\programs\python\python39\lib\site-packages (from -r requirements.txt (line 6)) (2021.1)
Collecting pycrypto>=2.6.1
  Using cached pycrypto-2.6.1.tar.gz (446 kB)
Collecting SungrowModbusTcpClient>=0.1.2
  Using cached SungrowModbusTcpClient-0.1.5-py3-none-any.whl (3.6 kB)
Requirement already satisfied: six>=1.15.0 in c:\users\administrator\appdata\local\programs\python\python39\lib\site-packages (from pymodbus>=1.3.2->-r requirements.txt (line 2)) (1.15.0)
Requirement already satisfied: pyserial>=3.4 in c:\users\administrator\appdata\local\programs\python\python39\lib\site-packages (from pymodbus>=1.3.2->-r requirements.txt (line 2)) (3.5)
Requirement already satisfied: requests<3,>=2 in c:\users\administrator\appdata\local\programs\python\python39\lib\site-packages (from dweepy>=0.3.0->-r requirements.txt (line 3)) (2.25.1)
Collecting msgpack
  Using cached msgpack-1.0.2-cp39-cp39-win_amd64.whl (68 kB)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in c:\users\administrator\appdata\local\programs\python\python39\lib\site-packages (from requests<3,>=2->dweepy>=0.3.0->-r requirements.txt (line 3)) (1.26.4)
Requirement already satisfied: chardet<5,>=3.0.2 in c:\users\administrator\appdata\local\programs\python\python39\lib\site-packages (from requests<3,>=2->dweepy>=0.3.0->-r requirements.txt (line 3)) (4.0.0)
Requirement already satisfied: idna<3,>=2.5 in c:\users\administrator\appdata\local\programs\python\python39\lib\site-packages (from requests<3,>=2->dweepy>=0.3.0->-r requirements.txt (line 3)) (2.10)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\administrator\appdata\local\programs\python\python39\lib\site-packages (from requests<3,>=2->dweepy>=0.3.0->-r requirements.txt (line 3)) (2020.12.5)
Using legacy 'setup.py install' for paho-mqtt, since package 'wheel' is not installed.
Using legacy 'setup.py install' for dweepy, since package 'wheel' is not installed.
Using legacy 'setup.py install' for pycrypto, since package 'wheel' is not installed.
Installing collected packages: pycrypto, msgpack, SungrowModbusTcpClient, paho-mqtt, influxdb, dweepy
    Running setup.py install for pycrypto ... error
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\administrator\appdata\local\programs\python\python39\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Administrator\\AppData\\Local\\Temp\\2\\pip-install-081hz3m7\\pycrypto_eefffd170dfb40a3b7fe2c7484cc70b6\\setup.py'"'"'; __file__='"'"'C:\\Users\\Administrator\\AppData\\Local\\Temp\\2\\pip-install-081hz3m7\\pycrypto_eefffd170dfb40a3b7fe2c7484cc70b6\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Administrator\AppData\Local\Temp\2\pip-record-dtko60m6\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\administrator\appdata\local\programs\python\python39\Include\pycrypto'
         cwd: C:\Users\Administrator\AppData\Local\Temp\2\pip-install-081hz3m7\pycrypto_eefffd170dfb40a3b7fe2c7484cc70b6\
    Complete output (183 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-3.9
    creating build\lib.win-amd64-3.9\Crypto
    copying lib\Crypto\pct_warnings.py -> build\lib.win-amd64-3.9\Crypto
    copying lib\Crypto\__init__.py -> build\lib.win-amd64-3.9\Crypto
    creating build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\hashalgo.py -> build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\HMAC.py -> build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\MD2.py -> build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\MD4.py -> build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\MD5.py -> build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\RIPEMD.py -> build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\SHA.py -> build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\SHA224.py -> build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\SHA256.py -> build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\SHA384.py -> build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\SHA512.py -> build\lib.win-amd64-3.9\Crypto\Hash
    copying lib\Crypto\Hash\__init__.py -> build\lib.win-amd64-3.9\Crypto\Hash
    creating build\lib.win-amd64-3.9\Crypto\Cipher
    copying lib\Crypto\Cipher\AES.py -> build\lib.win-amd64-3.9\Crypto\Cipher
    copying lib\Crypto\Cipher\ARC2.py -> build\lib.win-amd64-3.9\Crypto\Cipher
    copying lib\Crypto\Cipher\ARC4.py -> build\lib.win-amd64-3.9\Crypto\Cipher
    copying lib\Crypto\Cipher\blockalgo.py -> build\lib.win-amd64-3.9\Crypto\Cipher
    copying lib\Crypto\Cipher\Blowfish.py -> build\lib.win-amd64-3.9\Crypto\Cipher
    copying lib\Crypto\Cipher\CAST.py -> build\lib.win-amd64-3.9\Crypto\Cipher
[... 131 lines cut ...]
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(33): error C2059: syntax error: '}'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(35): error C2061: syntax error: identifier 'imaxdiv_t'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(35): error C2059: syntax error: ';'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(45): error C2143: syntax error: missing '{' before '__cdecl'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(46): error C2146: syntax error: missing ')' before identifier '_Number'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(46): error C2061: syntax error: identifier '_Number'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(46): error C2059: syntax error: ';'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(47): error C2059: syntax error: ')'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(50): error C2143: syntax error: missing '{' before '__cdecl'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(51): error C2146: syntax error: missing ')' before identifier '_Numerator'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(51): error C2061: syntax error: identifier '_Numerator'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(51): error C2059: syntax error: ';'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(51): error C2059: syntax error: ','
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(53): error C2059: syntax error: ')'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(55): error C2143: syntax error: missing '{' before '__cdecl'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt\inttypes.h(61): error C2143: syntax error: missing '{' before '__cdecl'
[... cut ...] _file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Administrator\AppData\Local\Temp\2\pip-record-dtko60m6\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\administrator\appdata\local\programs\python\python39\Include\pycrypto' Check the logs for full command output.

bdog720, 2021-04-29:
From what I've read pycrypto doesn't work anymore and people have suggested using pycryptodome

meltaxa, 2021-05-01:
Dependencies have been updated in the dev branch. Let me know if that helps and I'll merge.

bdog720, 2021-05-01:
Dev branch seems to work. However I messed around getting pycrypto to work (had to manually install it) so its hard to say if this solves it or not. Just a note that when its trying to install pycrypto it says VS c++ 14.0 build tools are required. Probably worth including that in the requirements?

bdog720, 2021-05-02:
Alright @meltaxa what I did was download/copy the SungrowModbusTcpClient folder from https://github.com/rpvelloso/Sungrow-Modbus into the Solariot folder.

I then modified the requirements text file of solariot. I removed SungrowModbusTcpClient>=0.1.5 and added pycryptodome

I'm sure there is a better way to do this as I have very little python / git experience

bdog720, 2021-05-02:
Ok @meltaxa this is what I did to fix it

git clone https://github.com/meltaxa/solariot.git --branch dev
git clone https://github.com/bdog720/Sungrow-Modbus --branch patch-1
cd Sungrow-Modbus
py setup.py install
cd ..
cd solariot
pip install -r requirements.txt

meltaxa, 2021-05-02:
Thanks for the feedback. I have cleaned up the requirements.txt in the dev branch. It relies on Sungrow-Modbus and will inherit it's dependencies. If your Sungrow-Modbus PR is accepted, it will flow on to Solariot.

Currently, I have tested successfully using the Docker solariot:dev image.

michael-robbins, 2021-05-02:
For what it's worth, the dev branch works for me too :D

meltaxa, 2021-05-04:
dev branch has been merged with master.

closed 2021-05-04 by meltaxa