meltaxa/solariot #40: Energy Generation on pvoutoput

shannonpasto, 2021-05-03:
For some reason Energy Generation (v1) has stopped updating today on pvoutoput. Everything else is updating (power, temp and voltage). The API call looks good so just wondering if anyone has any ideas? I've also posted on the forum.

`/service/r2/addstatus.jsp?d=20210503&t=14%3A40&c1=1&v1=3700.0&v2=1047&v5=40.6&v6=249.7 HTTP/1.1" 200 None`

Thanks

shannonpasto, 2021-05-03:
So I think there's something up with solariot API call. If I use the following command then Energy Generation uploads correctly..

`curl "https://pvoutput.org/service/r2/addstatus.jsp?sid=sid_key&key=api_key&d=20210504&t=08:20&v1=100&v2=239&v5=30.3&v6=245.2"`

michael-robbins, 2021-05-03:
Can you outline the 'difference' in API calls?
It could be the float/decimal vs the integer?

Also might be worth linking to the post over in the forums?

michael-robbins, 2021-05-03:
Having said that, my setup seems to be publishing daily and no real issues observed for v1 energy generation

shannonpasto, 2021-05-04:
I did a fresh git clone of the repo and reconfigured from scratch and it appears to be working. I've been watching it on and off today and so far so good but I have noticed something odd. Currently the telemetry is reporting daily_power_yield is 2400 and the API call is made with that value (`/service/r2/addstatus.jsp?d=20210504&t=13%3A40&c1=1&v1=2400.0&v2=314&v5=35.8&v6=247.8`) however the pvoutput site is only showing 2300. The last 3 are all showing the same. If I remove the c1 parameter then it updates as per the telemetry. I'm not sure I understand what the Cumulative Flag is all about?

michael-robbins, 2021-05-04:
Guessing this is your post: https://forum.pvoutput.org/t/energy-generated-not-updating/4304/2

Going off the response in there, the number has to 'increase' for it to be recorded, otherwise it'll be reported as 0...

And the issue here is that you're reporting '2300' at multiple status updates, which breaks the 'has to increase on every status update' logic I'm guessing? (just thinking out loud here)

shannonpasto, 2021-05-04:
Yeah, that's it. Didn't see the response.

The thing is daily_power_yield was increasing earlier in the day according to the telemetry but that was not being reflected on pvoutput. What I don't get is that from 28/04 to 02/05 this was working perfectly. Something happened (overnight) on 03/05 

If daily_power_yield is 2400 at 13:40 then 2500 at 13:45 then pvoutput should reflect that but from what I was seeing today that was not the case when `c1=1`. What I might do is add `c1=1` back in tomorrow and see what happens.

shannonpasto, 2021-05-04:
Ok, so the original issue of Energy Generation not updating is resolved by a fresh clone of the repo so I'm ok to close this. I am seeing some weird behaviour with the `c1=1` flag but it is increasing albeit a few API calls behind so I'm going to assume that pvoutpout is doing some sort of calculations over time for other pieces of data, eg efficiency, average etc. This is something I can raise on pvoutoput to get a better understanding of this flag.

I think I may have found a small bug with the writing of the telemetry.json file though so I'll raise a separate issue when I have a moment in the next day or so.

@michael-robbins thanks for your help

closed 2021-05-04 by shannonpasto