meltaxa/solariot #44: Home Assistant - MQTT Broker - Socket Error

tgcowell, 2021-05-19:
Hi There,

I'm helping a mate connect his Solar Invertor to HA via MQTT, I've setup Solariot via Docker and can confirm it's connected and can see in the logs it's pulling data from the invertor etc.

However it's pushing to MQTT broker however it's claiming there is a socket error

1621386684: New client connected from 192.168.1.111 as pv_data (p2, c1, k60, u'xxxxxx').
1621386695: Socket error on client pv_data, disconnecting

googling it looks like it's an error within the solariot code but I'm not really clear if this is correct? something about a unique ID.

Any advice would be appreciated.

michael-robbins, 2021-05-29:
What MQTT broker are you using?
What are the settings on the MQTT broker? Do you have authentication enabled? allow anonymous auth?

tgcowell, 2021-05-29:
Thank you for your reply,

I'm using the mosquito broker, this is built into the Hass.io supervisor store. Authentication is turned on although TLS is not.

I'm not sure if anonymous auth is activated, will review and confirm.

While the system 100% was pulling data, it would only last about 10 minutes before an error would crash. We are accessing via a wifi dongle on the same SG5K-D so I'm curious if eclipse is different in some way? I'll see if we can set this up as well and test just for completeness. 

Most people can't get the TLS working on the mosquito one via HA so that would be a nice feature if available via eclipse.

michael-robbins, 2021-05-30:
Ohh ok so it's actually working end-to-end, just after some time there's error(s).

If you've got a user/pass setup then anon auth isn't needed.

I've got just a separate mqtt broker running with Traefik infront providing (transparent) TLS, mqtt broker itself thinks there's no TLS.

It seems to be working just fine, I've got hass as well as mqtt2prometheus listening to the solariot topic and they're both fine.

What's the crash? Do you have any logs? I suspect this is probably a hass mqtt broker problem though and not solariot?

tgcowell, 2021-09-08:
Exception in thread Thread-293:                                                                                                                                                                                                                                                                                                                                                                           
Traceback (most recent call last):                                                                                                                                                                                                                                                                                                                                                                        
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner                                                                                                                                                                                                                                                                                                                             
    self.run()                                                                                                                                                                                                                                                                                                                                                                                            
  File "/usr/local/lib/python3.9/threading.py", line 910, in run                                                                                                                                                                                                                                                                                                                                          
    self._target(*self._args, **self._kwargs)                                                                                                                                                                                                                                                                                                                                                             
  File "/solariot/solariot.py", line 402, in publish_mqtt                                                                                                                                                                                                                                                                                                                                                 
    mqtt_client.reconnect()                                                                                                                                                                                                                                                                                                                                                                               
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1075, in reconnect                                                                                                                                                                                                                                                                                                              
    sock = self._create_socket_connection()                                                                                                                                                                                                                                                                                                                                                               
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3546, in _create_socket_connection                                                                                                                                                                                                                                                                                              
    return socket.create_connection(addr, source_address=source, timeout=self._keepalive)                                                                                                                                                                                                                                                                                                                 
  File "/usr/local/lib/python3.9/socket.py", line 844, in create_connection                                                                                                                                                                                                                                                                                                                               
    raise err                                                                                                                                                                                                                                                                                                                                                                 [... cut ...]
    sock.connect(sa)                                                                                                                                                                                                                                                                                                                                                                                      
socket.timeout: timed out      

Please note I've found this in the logs... so it will connect and grab the data, followed by the above error.

tgcowell, 2021-09-08:
I will add this is being run via a docker... not sure if there is anything specific needed for this to work? albeit I am connecting to the invertor and grabbing the data.

michael-robbins, 2021-09-08:
I guess that error above is more to do with the MQTT server not being availabe/routable/connectable from solariot..?

If it cannot create the socket, that just means roughly "the packets never made it to MQTT" or "the packets never made it back from MQTT" and doesn't seem to be a solariot specific issue

michael-robbins, 2021-09-08:
If it's in docker can you `exec` into the solariot container and `telnet` to the mqtt IP/port to verify you can manually connect?

tgcowell, 2021-09-08:
root@meltaxa-solariot1:/solariot# telnet                                                                                                                    
telnet> open 192.168.1.222 1883                                                                                                                             
Trying 192.168.1.222...                                                                                                                                     
Connected to 192.168.1.222.                                                                                                                                 

Yes, I can connect via Telnet and can also confirm the broker is seeing the connection

1631067992: Socket error on client pv_data, disconnecting.
1631067992: New connection from 192.168.1.111 on port 1883.
1631067992: New client connected from 192.168.1.111 as pv_data (p2, c1, k60, u'mqttuser').
1631068003: Socket error on client pv_data, disconnecting.
1631068003: New connection from 192.168.1.111 on port 1883.
1631068003: New client connected from 192.168.1.111 as pv_data (p2, c1, k60, u'mqttuser').
1631068004: Client <unknown> has exceeded timeout, disconnecting.
1631068015: Socket error on client pv_data, disconnecting.
1631068015: New connection from 192.168.1.111 on port 1883.
1631068015: New client connected from 192.168.1.111 as pv_data (p2, c1, k60, u'mqttuser').
1631068026: Socket error on client pv_data, disconnecting.
1631068026: New connection from 192.168.1.111 on port 1883.
1631068026: New client connected from 192.168.1.111 as pv_data (p2, c1, k60, u'mqttuser').
1631068038: Socket error on client pv_data, disconnecting.
1631068038: New connection from 192.168.1.111 on port 1883.
1631068038: New client connected from 192.168.1.111 as pv_data (p2, c1, k60, u'mqttuser').
1631068049: Socket error on client pv_data, disconnecting.

I just keep getting this socket error message can't work out what it is referring too???

tgcowell, 2021-09-08:
Message 0 received on inverter/stats at 12:49:
{
    "daily_power_yield": 19500,
    "total_power_yield": 73.93,
    "internal_temp": 51.2,
    "pv1_voltage": 296.9,
    "pv1_current": 9.3,
    "pv2_voltage": 190.8,
    "pv2_current": 8.4,
    "total_pv_power": 4382,
    "grid_voltage": 250.3,
    "inverter_current": 16.4,
    "total_active_power": 4382,
    "grid_frequency": 50,
    "export_power": -4381,
    "export_power_indicator": 65535,
    "power_meter": 0,
    "daily_purchased_energy": 13.2,
    "daily_energy_consumption": 3400,
    "total_energy_consumption": 1695.8,
    "year": 2021,
    "month": 9,
    "day": 8,
    "hour": 12,
    "minute": 49,
    "second": 13,
    "timestamp": "8/9/2021 12:49:13"
}
QoS: 0 - Retain: false

I can also confirm if I listen for  inverter/stats the broker receives the message... it's just something to do with socket error on client pv_data ... I have no idea what this could mean... any help is appreciated.