meltaxa/solariot #45: unable to decode modbus response

vikram1505, 2021-05-26:
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 54241)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 56349)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 42925)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 48907)
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 53489)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 41505)
DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x88 0x0 0x64
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Transaction failed. ([Errno 104] Connection reset by peer)
DEBUG:pymodbus.framer.socket_framer:Processing:
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
WARNING:root:Modbus connection failed
WARNING:root:Failed to scrape inverter, sleeping until next scan
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 54319)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 40039)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 37303)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 59533)
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 50211)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.250', 46091)
DEBUG:pymodbus.transaction:SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x88 0x0 0x64
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x0
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0
DEBUG:pymodbus.factory:Factory Response[0]
ERROR:pymodbus.factory:Unable to decode response Modbus Error: Unknown response 0
ERROR:pymodbus.transaction:Modbus Error: [Input/Output] Unable to decode request

slackerlinux85, 2021-05-28:
this might be related to pymodbus 2.5.1

what inverter do you have i have a Sungrow SG3k-S that does the same thing with pymodbus >=2.5.1(maybe 2.5.0 havent tested it) you can downgrade pymodbus by doing `pip install pymodbus==2.4.0`

it could also be you just haven't waited enough time the sungrow inverters always seem to reset by peer then a 0 response then real data as shown in logs below too.

attached is a log where i downgrade pymodbus to 2.4.0 and show the script working and then update to latest and show it broken.(this happens on multiple systems and OS. slackware64 and ubuntu-server 20.04(arm64 and x86_64) tested)

[pymodbuslog.log](https://github.com/meltaxa/solariot/files/6558157/pymodbuslog.log)

vikram1505, 2021-05-30:
Thank you so much for your help!

downgrading fixed the issue instantly, solariot is not exporting data to mqtt, however now the next problem is how to get that data from mqtt and add a sensor to  homeassistant i.e. extraction of data from mqtt.

slackerlinux85, 2021-05-30:
would be good to know some info on your setup to see what HW/FW might be affected. mine is SG3K-S with a wifi module V312 installed in march(so fairly recent HW/FW). ill add firmware versions later(dont have their app installed atm and also have to unblock it from router)

vikram1505, 2021-05-30:
inverter is SG5K-D, wifi WiFi V31_001_247, I am running solariot on one raspberry pi which is feed to mqtt running as an add on another raspberry pi with Home assistant

wapayne69, 2021-09-01:
This still appears to be an issue. I had to downgrade to pymodbus==2.4.0 in to get solariot.py to successfully run on my raspberry pi (Its scraping data from my sg8k invertor). This solves the problem when running the python script but doesn't help much in terms of running the docker image...

meltaxa, 2021-09-01:
Change the requirements.txt to lock in that pymodbus version and redeploy the docker container. Let me know how you go and I'll make the updates in the repo.

wapayne69, 2021-09-01:
> Change the requirements.txt to lock in that pymodbus version and redeploy the docker container. Let me know how you go and I'll make the updates in the repo. Note: you'll need to build the image locally to test (as opposed to pulling down from the hub).

Thanks, yeah it works when building the image on my side.

meltaxa, 2021-09-03:
Works for me too. Merged dev -> master.

closed 2021-09-03 by meltaxa

vikram1505, 2021-09-15:
wrote a wrapper to convert deamonise this in systemd, this can now look for when the inverter online and then poll it , stops polling when the inverter goes offline, mimics following sunrise to sunset, so far has been working as expected.

PS I am not a professional developer/programmer so apologies if I have posted this at a wrong location, but i guess as the wrapper is around the original "solariot" project this should still be the right place, also i have never contributed to a opensource community this is my first time so please provide your feed back

#!/bin/bash

# check if inverter is alive

function alive_check () {
        count=$( ping -c 1 172.28.25.11 | grep icmp* | wc -l )
        target=172.28.25.11
        check="Down"

        if ping -c 1 -W 1 "$target" ; then
                check="Up"
        fi
        echo $check

}

##Start Polling using solariot

function startwrapper () {
        cd /home/vikram/solariot/
        logs=/home/vikram/solariot/logs.txt
        echo "`date` >> START ==> Invoked " >> $logs
        echo "`date` >> START ==> current path == `pwd` " >> $logs
        check=`ps xauuuf | grep solariot.py | grep python3 | awk '{print $2}' | sed 's/ //g'`
        while [ -z "$check" ]; do
                echo "`date` >> STARTUP ==> solariot not running starting now" >> $logs
                ./solariot.py -vv &
               sleep 30
               check=`ps xauuuf | grep solariot.py | grep python3 | awk '{print $2}' | sed 's/ //g'`
        done

        if [ -z "$check" ] ; then
                echo "`date` >> STARTUP ==> failed to start solariot " >> $logs
        else
                echo "`date` >> STARTUP ==> solariot started successfully PID = $check" >> $logs
        fi
}

## checks if solariot is running
function collector_check (){
        cd /home/vikram/solariot/
        logs=/home/vikram/solariot/logs.txt
        findpid=`ps xauuuf | grep solariot.py | grep python3 | awk '{print $2}' | sed 's/ //g'`
        pidcheck="Stopped"
        if [ ! -z "$findpid" ]; then
                pidcheck="Running"
        fi
}

## Stops solariot
function stopwrapper () {

        cd /home/vikram/solariot/
        logs=/home/vikram/solariot/logs.txt
        check=`ps xauuuf | grep solariot.py | grep python3 | awk '{print $2}' | sed 's/ //g'`
        if [ ! -z "$check" ]  ; then
                echo "`date` stopwrapper ==>  solariot running PID == $check sending kill signal" >> $logs
                kill $check
                sleep 5
                check=`ps xauuuf | grep solariot.py | grep python3 | awk '{print $2}' | sed 's/ //g'`
                if [ -z "$check" ] ; then
                        echo "`date` stopwrapper ==>  solariot stopped successfully " >> $logs
                        exit 127
                else
                        echo "`date` stopwrapper ==>  failed to stop solariot please check manually PID == $check" >> $logs
                fi
        else
                echo "`date` stopwrapper ==>  Solariot not running " >> $logs
        fi
}

###MAIN here 

echo "`date` >> SolarMonitor.Service >> Daemon Running" >> /home/vikram/solariot/logs.txt
while true; do
        sleep 30
        alive_check
        #echo $check
        if [ "$check" = "Up" ]; then
                collector_check
                #echo "$pidcheck >> $findpid"
                if [ "$pidcheck" = "Running" ]; then
                        continue
                else
                        echo "`date` >> SolarMonitor.Service >> Inverter Alive will start polling it " >> /home/vikram/solariot/logs.txt
                        startwrapper
                        echo "`date` >> SolarMonitor.Service >> Polling Started " >> /home/vikram/solariot/logs.txt
                fi
        else
                collector_check
                if [ "$pidcheck" = "Stopped" ]; then
                        continue
                else
                        echo "`date` >> SolarMonitor.Service >> Inverter is Offline! Stopping Poller now " >> /home/vikram/solariot/logs.txt
                        stopwrapper
                        echo "`date` >> SolarMonitor.Service >> Polling stopped, will wait for inverter to be available again " >> /home/vikram/solariot/logs.txt
                fi
        fi

        sleep 60
done

meltaxa, 2021-09-15:
Hi @vikram1505 thanks for your comment. 

I try to keep solariot platform agnostic. Note, systemd is not available on Windows and not all Linux systems too, such as OSX. 

Although, I'm sure other systemd Linux users would find your input of value, I would go with a different approach. Perhaps update the actual solariot.py code to sleep when the inverter is "offline" and rely on the Docker container for a "daemonized" approach. If you wish to pursue either approach, fork the project, make your changes (don't hard code IP addresses or paths) and submit a PR.

vikram1505, 2021-09-15:
Thanks Meltaxa, do you recon coming with a config file approach for inverter IP and paths ? Sorry I am little new to this stupid questions how to fork the project ? How to submit a PR ?

meltaxa, 2021-09-15:
Correct, glean the IP address from the config file and the path from the user environment variables. 

As for forking and PR's, you may need to step back a bit and get a good handle on using git (clone, branch, commit, pull, push) commands. Not easy to cover here. Best to visit Github's own docs (links in the footer) on this subject or even Atlassian's git guides. A PR will be critiqued and others may offer guidance on your code quality and also suggested methods to achieve the same. Don't take those reviews as personal :-)

EvoSems, 2024-01-31:
> this might be related to pymodbus 2.5.1
>
> what inverter do you have i have a Sungrow SG3k-S that does the same thing with pymodbus >=2.5.1(maybe 2.5.0 havent tested it) you can downgrade pymodbus by [...]

how can I downgrade this in a rip docker container? thanks