meltaxa/solariot #48: sungrow-sg5kd Modbus connection failed

Flynsarmy, 2021-06-23:
I can see ports 80 and 502 open and can telnet into port 502 fine to confirm a connection can be made but solariot is giving a connection failed. Connecting via wifi. I've tried slave 0x01 through 0x04.

`cat config.py`
> inverter_ip = "10.0.0.47"
> inverter_port = 502
> slave = 0x01
> model = "sungrow-sg5kd"
> timeout = 3
> scan_interval = 10
> dweepy_uuid = "random-uuid"

Returns:
`./solariot.py -vv`
> INFO:root:Loaded config sungrow-sg5kd
> INFO:root:Creating SungrowModbusTcpClient
> INFO:root:Connecting
> DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('10.0.0.26', 53763)
> DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('10.0.0.26', 53764)
> INFO:root:Connected
> INFO:root:No MQTT configuration detected
> INFO:root:No InfluxDB configuration detected
> INFO:root:No PVOutput configuration detected
> DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('10.0.0.26', 53765)
> DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('10.0.0.26', 53766)
> DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('10.0.0.26', 53767)
> DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('10.0.0.26', 53768)
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('10.0.0.26', 53769)
> DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('10.0.0.26', 53770)
> DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x88 0x0 0x64
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Transaction failed. ([Errno 54] Connection reset by peer)
> DEBUG:pymodbus.framer.socket_framer:Processing:
> DEBUG:pymodbus.transaction:Getting transaction 1
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> WARNING:root:Modbus connection failed
> WARNING:root:Failed to scrape inverter, sleeping until next scan

mhoroschun, 2021-07-03:
Sounds like the same problem as #45? See also my discussion here #34.

Flynsarmy, 2021-07-06:
That got it. TYVM!

closed 2021-07-06 by Flynsarmy