meltaxa/solariot #49: Sungrow SG10RT inverter modbus

rark-ha, 2021-07-04:
Hi. I've recently installed a SG10RT inverter (3 Phase) and trying to work out the modbus setup.

If I use the model = "sungrow-sg5kd" I can get most of the readings, except "power consumption" and "energy consumption".

(sungrow-sh5k modbus fails to connect at all).

Can someone point me in the right direction to set up a modbus file that collects the power and energy info to?

rark-ha, 2021-07-06:
I can scrape the 5000-5100 range and the 7000-7100 (though not much there) using function 04. I can't get the 13000-13100 range to scrape with function 04 as it says illegal data address.

sungrow-sh5k pulls the values I'm still missing from 13000 range and nothing in the 5000 range appears to be it.

STL2020, 2021-07-06:
<html>
<body>

2021-07-06 10:29:39 | stdout | pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(192.168.2.93:502): Connection unexpectedly closed 1.477375 seconds into read of 25 bytes without response from unit before it closed connection
-- | -- | --
2021-07-06 10:29:39 | stdout | raise ConnectionException(msg)
2021-07-06 10:29:39 | stdout | File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 343, in _handle_abrupt_socket_close
2021-07-06 10:29:39 | stdout | return self._handle_abrupt_socket_close(
2021-07-06 10:29:39 | stdout | File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 298, in _recv
2021-07-06 10:29:39 | stdout | self._key_packet = self._recv(25)
2021-07-06 10:29:39 | stdout | File "/usr/local/lib/python3.9/site-packages/SungrowModbusTcpClient/SungrowModbusTcpClient.py", line 40, in _getkey
2021-07-06 10:29:39 | stdout | self._getkey()
2021-07-06 10:29:39 | stdout | File "/usr/local/lib/python3.9/site-packages/SungrowModbusTcpClient/SungrowModbusTcpClient.py", line 54, in connect
2021-07-06 10:29:39 | stdout | client.connect()
2021-07-06 10:29:39 | stdout | File "/solariot/solariot.py", line 108, in <module>
2021-07-06 10:29:39 | stdout | Traceback (most recent call last):
2021-07-06 10:29:38 | stdout | INFO:root:Connecting
2021-07-06 10:29:38 | stdout | INFO:root:Creating SungrowModbusTcpClient
2021-07-06 10:29:38 | stdout | INFO:root:Loaded config sungrow-sg5kd

</body>
</html>

STL2020, 2021-07-06:
The same  Problem with modbus TCP .No connect

rark-ha, 2021-07-07:
Further investigation from the WiNet WiFi module local web output shows real time data and also has a read back function for the modbus data (but only in hexadecimal).

(Not being an expert my modbus address shows out by 1 compared with solariot files e.g. internal_temp_10 for my scan is 5007 but maps fine with 5008 in solariot).

5002 daily_power_yield_0.01 (5003 solariot)
5003 total_power_yield_100 _(5004 solariot)_
5005 #NEW Total On-Grid Running Time (hrs)
5007 internal_temp_10 _(5008 solariot)_
5008 #NEW Apparent Power_0.001 (kVa)
5010 pv1_voltage_10 (5011 solariot) (called MPPT1 Voltage in web interface)
5011 pv1_current_10 (5012 solariot) (called MPPT1 Current in web interface)
5012 pv2_voltage_10 (5013 solariot) (called MPPT2 Voltage in web interface)
5013 pv2_current_10 (5014 solariot) (called MPPT2 Current in web interface)
5016 #NEW Total DC Power 
5018 #NEW Phase A Voltage_0.1
5019 #NEW Phase B Voltage_0.1
5020 #NEW Phase C Voltage_0.1
5021 #NEW Phase A Current_0.1
5022 #NEW Phase B Current_0.1
5023 #NEW Phase C Current_0.1
5030 total_active_power (5031 solariot)
5035 grid_frequency_10 (5036 solariot)
5070 #NEW Array Insulation Resistance_10 (Kohms)

TBA
5037 shows values when exporting (as large negative, otherwise seems to be 0)

If I scan 5100+125 range I come up with further values but trying to determine what they are. Those that don't seem to run up sequentially when logging are;
5215 potentially an export value (negative when exporting, but smaller positive numbers other times)
5216 potentially an export flag (-1 when exporting, 0 when not)
5217 potentially consumption

Any help deciphering the logic etc would be appreciated. Thinking of sending all these values through to InfluxDB if I can modify the modbus file and allow easier logging and value graphing.

bohdan-s, 2021-09-26:
Hi, I have the official modbus map from Sungrow. 
I dont want to post it here, but can send you a copy if you want.

bohdan-s, 2021-09-26:
daily_purchased_energy_10 = "Daily import energy"? 5097~5098 U32
    daily_energy_consumption_0.01 = "Daily direct energy consumption"? 5101~5102 U32
    total_energy_consumption_10 = "Total direct energy consumption"? 5103~5104 U32

meltaxa, 2021-09-26:
Hi @bohdan-s. See my Github bio for my contact details to send me a copy. Cheers!

freman, 2021-10-15:
Hi @bohdan-s, I also would be super interested if I may have a copy? I'm trying to get the Sungrow to work along side my Fronius (fronius is doing the solar, sungrow is doing the battery) and I'm really just trying to fill in the blanks

freman, 2021-10-18:
@bohdan-s no worries, and thanks. this time was used figuring out modbus, how it works, and what it means. I've learned a lot so now I can actually use this knowledge :D