meltaxa/solariot #53: WiFi vs Ethernet

rark-ha, 2021-07-24:
I had my inverter connected via Wi-Fi until I had time to run an ethernet cable to it. Switched it over to an ethernet connection and iCloudsolar etc continued to work without issue.

But solaroit.py kept reporting 'Modbus connection failed' and 'Failed to scrape inverter, sleeping until next scan' in the log file.

I'd updated the config.py to the ethernet ip address but it didn't work still. I could run a Modbus poll from another machine without any issue.

Also tried;
- Restarting the docker container
- Resetting and starting again with the docker container
- Turning off WiFi via the local web interface for the inverter
- Moving the ethernet IP back to the original WiFi one (and update config.py)
- I could ping and telnet into the ip/port from the docker machine
- If I changed IP (without updating the config.py) the log would give me 'Connection to (192.168.20.42, 502) failed: [Errno 111] Connection refused' so it was connecting to the ip address previously

I turned wifi back on (and ethernet still on), pointed config.py to the wifi ip, and it started working again.

Why the issue with ethernet?

rark-ha, 2021-08-09:
I've discovered that I now cannot connect to the modbus. I've tried it from a few different systems but it won't read the input registers (e.g. 5000 + 1000) as it returns illegal address details. The only thing I can pull is the holding registers 4999 + 10 (so I know it's connecting).

From the inverter web interface I can read these input registers from the testing page so I know the address is correct.

Any ideas?

rark-ha, 2021-08-10:
This is the error I'm now getting trying to read the input registers.

`
ERROR:pymodbus.factory:index out of range | stdout
ERROR:pymodbus.transaction:Modbus Error: [Input/Output] Unable to decode request
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
`

meltaxa, 2021-09-03:
Try the latest code which addresses the pymodbus version issue.

rark-ha, 2021-09-04:
Still get the same error.

bohdan-s, 2021-09-26:
Hi,
I am getting the same issues. New install of a Sungrow SG7.0RT. 
I have the modbus map from Sungrow so know I am hitting the right registers.
I can read the holding registers OK and run it from the web interface.
I have moved from WiFi to LAN and still same issue.

bohdan-s, 2021-09-26:
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64489)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64490)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64491)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64492)
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 13
DEBUG:pymodbus.transaction:Clearing current Frame : - 0x0 0xc 0x0 0x0 0x0 0x2 0x1 0x84 0x2
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64493)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64494)
DEBUG:pymodbus.transaction:SEND: 0x0 0xd 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x88 0x0 0x64
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x0 0xd 0x0 0x0 0x0 0x2 0x1 0x84 0x2
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0xd 0x0 0x0 0x0 0x2 0x1 0x84 0x2
DEBUG:pymodbus.factory:Factory Response[132]
ERROR:pymodbus.factory:index out of range
ERROR:pymodbus.transaction:Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
WARNING:root:Modbus connection failed
WARNING:root:Failed to scrape inverter, sleeping until next scan
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64520)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64521)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64522)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64523)
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 14
DEBUG:pymodbus.transaction:Clearing current Frame : - 0x0 0xd 0x0 0x0 0x0 0x2 0x1 0x84 0x2
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64524)
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.1.148', 64525)
DEBUG:pymodbus.transaction:SEND: 0x0 0xe 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x88 0x0 0x64
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x0 0xe 0x0 0x0 0x0 0x2 0x1 0x84 0x2
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0xe 0x0 0x0 0x0 0x2 0x1 0x84 0x2
DEBUG:pymodbus.factory:Factory Response[132]
ERROR:pymodbus.factory:index out of range
ERROR:pymodbus.transaction:Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
WARNING:root:Modbus connection failed
WARNING:root:Failed to scrape inverter, sleeping until next scan

rark-ha, 2021-10-21:
> I can read the holding registers OK and run it from the web interface. I have moved from WiFi to LAN and still same issue.

This is my exact issue. Perhaps coincidental that it was working and stopped when I went to LAN, but it doesn't matter if I remove LAN now or not. I have the same error log file as you.

bohdan-s, 2021-12-06:
Hi, please see #66, I have created a new library SungrowModbusWebClient to pull the modbus via HTTP requests. It's dirty but I have been running it now for a week.

closed 2022-01-26 by meltaxa