meltaxa/solariot #70: SG15RT support

lukecarbis, 2022-01-20:
It would be great if this supported the SG15RT inverter.

michael-robbins, 2022-02-05:
Unless someone else has this inverter and willing to give it a shot, you're going to have to 'have a go yourself' as only then can the registers be confirmed and a mapping file be produced :)

You can always copy a mapping file from another (similar) inverter and start from there! That's what I did!

lucahammer, 2022-04-07:
I have a SG20RT with a WiNet-S dongle. From a first test, solariot works without issues when using 'sungrow-sg10rt' as model in the config. I assume that the modbus registers of SG5.0RT, SG7.0RT, SG10RT, SG15RT, SG20RT are the same.

Sample output from my test:
```{'daily_power_yield': 7600.0,
 'total_power_yield': 2.04,
 'total_ongrid_running_time': 82,
 'internal_temp': 42.5,
 'apparent_power': 1675000.0,
 'pv1_voltage': 592.1,
 'pv1_current': 1.5,
 'pv2_voltage': 577.3,
 'pv2_current': 1.4,
 'total_pv_power': 1696,
 'grid_voltage_a': 232.8,
 'grid_voltage_b': 232.8,
 'grid_voltage_c': 230.4,
 'inverter_current_a': 2.4,
 'inverter_current_b': 2.4,
 'inverter_current_c': 2.4,
 'total_active_power': 1675,
 'grid_frequency': 499.8,
 'array_insulation_resistance': 318.0,
 'daily_operation_time': 237,
 'monthly_power_yield': 200500.0,
 'export_power': 0,
 'export_power_indicator': 65535,
 'power_meter': 65535,
 'year': 2022,
 'month': 4,
 'day': 7,
 'hour': 12,
 'minute': 17,
 'second': 10}```

closed 2023-02-07 by lukecarbis