meltaxa/solariot #71: Command errored out with exit status 1

sEdsKi, 2022-01-23:
Hi, Sorry - I have next to no idea what I'm doing.

I've attempted to install this on a Linux Virtual Machine. Is that possible?
I got to this point... 

[image]

breed808, 2022-01-23:
Do you have a C compiler installed? It's necessary to build the cryptography library used by `solariot`.

I can't find any mention of installing system packages in the official documentation, but [Docker is apparently installed](https://github.com/home-assistant/operating-system). You could attempt to run `solariot` via the Docker container.

sEdsKi, 2022-01-25:
> Do you have a C compiler installed?

Not unless it was installed along with the HAOS for Virtual Machine. 
I have tried searching and reading up - but cannot find anything that looks suitable

> You could attempt to run `solariot` via the Docker container.

I'm not sure how to go about this

meltaxa, 2022-01-26:
Try doing the same in a fresh python virtual environment. For example:
```
python3 -m venv solariot-venv
cd solariot-venv
source bin/activate
git clone https://github.com/meltaxa/solariot.git
cd solariot
pip install --upgrade -r requirements.txt
```
I was able to do the above without error (on Python 3.8).

As mentioned, Docker is also another way to run Solariot. You'll need to install Docker Desktop: https://www.docker.com/products/docker-desktop. Follow the instructions under the Docker heading in the readme.

sEdsKi, 2022-01-28:
Both of these options are beyond my full understanding.
Should I be running these commands through ssh?  Some other method?