meltaxa/solariot #79: Error loading config.py

jeremyw24, 2022-09-05:
I had everything setup and running fine for a few days.

Then it stopped logging, following investigation I found the issue:
```
pi@homebridge:~/solariot $ python3 --version
Python 3.5.2
pi@homebridge:~/solariot $ python3 ./solariot.py
  File "./solariot.py", line 69
    logging.info(f"Loaded config {config.model}")
                                               ^
SyntaxError: invalid syntax
```

I am not really familiar with Python but have had a poke around and can't seem to figure out what the issue is, any assistance would be appreciated?

breed808, 2022-09-05:
The [f-strings](https://peps.python.org/pep-0498/) interpolation used here was [introduced in Python 3.6](https://docs.python.org/3/whatsnew/3.6.html#summary-release-highlights).
You'll need to use Python 3.6 or newer when running this tool.

jeremyw24, 2022-09-06:
Thanks @breed808 Ill update it.

closed 2022-09-06 by jeremyw24