meltaxa/solariot #104: solariot errors when run

iankds05, 2024-10-06:
I have installed this repository on Ubuntu desktop and followed the install procedure as outlined in the readme. Everything seems to be installed correct and I have created my own inverter model file SG10RS using the provided SG10rt file but with applicable modbus addresses plus made the config.py file with mqtt settings When the solarart.py is run from terminal I get the following errors

Sungrow@Ubuntu:~/solariot$ ./solariot.py
Traceback (most recent call last):
  File "/home/Sungrow/solariot/./solariot.py", line 127, in <module>
    mqtt_client = mqtt.Client(getattr(config, "mqtt_client_name", "pv_data"))
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Sungrow/.local/lib/python3.12/site-packages/paho/mqtt/client.py", line 772, in __init__
    raise ValueError(
ValueError: Unsupported callback API version: version 2.0 added a callback_api_version, see docs/migrations.rst for details

As I am no way skilled at python could you maybe give me some pointers as to what could be the possible issue 
Thanks

TOMSLAUS, 2024-11-10:
Currently looking to use SolarIot, wondering if you managed to fix the issue?

kyliejudd, 2024-11-26:
If you are lazy like me, you can downgrade paho-mqtt, 
`pip install "paho-mqtt<2.0"`
It looks like it was a breaking change in 2.0
https://eclipse.dev/paho/files/paho.mqtt.python/html/migrations.html

zyrorl, 2026-02-14:
I've fixed this in a new PR, i'm hoping it'll get merged in.